F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #5: Holiday Mode Doesn't Stay Active on Cascaded Ecodan

drnichols, 2024-09-17:
Hi @F1p, not sure if this is just an Ecodan cascade oddity...when I enable holiday mode it stays on for a few seconds before automatically reverting to disabled. It does the same from the main controller so it may well be a cascade 'feature'. Holiday mode worked ok when my systems were standalone. My systems are all running different FW which could also be a factor (I'm hoping to get them updated to the same version). Any thoughts or debugging I can do?

F1p, 2024-09-17:
I found my FTC5 actually does the same, it does latch but then quickly unlatches - i wondered if it was my FTC firmware

drnichols, 2024-09-18:
@F1p I'm experimenting with server control now which I think will replace the Holiday Mode for me going forwards, plus opens up many more automation options. Nice work!

F1p, 2024-11-26:
Holiday mode not latching is now fixed!