F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #8: Merge Binaries for Flashing

lboue, 2024-10-04:
Hello,

Would it be possible to supply a single binary directly using the following command? This would simplify the flashing operation:

```
esptool.py --chip esp32s3 merge_bin -o merged-flash.bin --flash_mode dio --flash_size 4MB 0x0000 ECODAN_Bridge_v5.2.0.bootloader.bin 0x8000 ECODAN_Bridge_v5.2.0.partitions.bin 0x10000 ECODAN_Bridge_v5.2.0.bin
```

Doc: [Merge Binaries for Flashing: merge_bin](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html#merge-binaries-for-flashing-merge-bin)

> The merge_bin command will merge multiple binary files (of any kind) into a single file that can be flashed to a device later. Any gaps between the input files are padded based on the selected output format.

F1p, 2024-10-04:
Merged binary is only required for first flash, then you can use OTA and flash just the Vx.x.x.bin file

I have provided it [here](https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/tree/master/build/esp32.esp32.m5stack-atoms3/v5.2.6)

Note, it is easier when moved to Core 3.0.5/IDF5.4, this build has a bug in the LED control and it is awaiting fix for issue [FastLED](https://github.com/FastLED/FastLED/issues/1631)

F1p, 2024-10-04:
Sorry i had to revert 5.2.6
The issue linked above is still causing problems - once it is resolved i can publish merged builds

F1p, 2024-11-26:
Now implemented