F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #10: Server Control Module - turning off when running DHW Boost

merlin051, 2024-12-03:
Ecodan 11.2kw
FTC6 v19
Firmware 5.3.1

Server control is up and running, but when DHW Boost is manually stopped or just finishes it's cycle the Server Control is deactivated.

I can automate this turning back on as a work around but I don't think it's supposed to happen.

F1p, 2024-12-03:
The solution is when using server control mode at all times is to turn off prohibit DHW when you wish to boost, then a second automation to re-enable when the System Operation Mode changes from Hot Water (as per example automation).

The use of DHW boost (not fast) was aimed at users who didn't wish to leave the unit in server control mode at all times. This function could be changed if you were already in server control mode

merlin051, 2024-12-03:
Is this documented somewhere?

I did see the automation and wonder if the DHW whilst in server control is in 'always on' mode using the min/max/temp drop values? (But wasn't sure hence going with the scheduled boost)

Guessing from your comment the Fast DHW option is something else and can be run in server control? (I assumed this runs the immersion too)

F1p, 2024-12-03:
Fast DHW is the same as 'boost now' from the main controller or remote thermostats, it will run the compressor at a high frequency and ignore DeltaT to being the temperature up as rapidly as possible - it may use immersion to assist at higher setpoints

I think I would rather resolve this with returning server control mode to the state it begins from when it's triggered

merlin051, 2024-12-03:
I think I'm just unsure of best practice/design from the server control perspective, everything I had configured previously I was doing via MelCloud plugin or the controller timers (DHW ran at 03:30 regardless of temperature to top the tank up for the next day) so maybe it's not a direct lift and shift.

Learning curve maybe?

F1p, 2024-12-03:
Yes it's a learning curve, and not nessesary for everyone.
The majority of server control users are on an Agile tarriff, where they use it to prohibit heating everything between 4pm and 7pm

In addition, they may have a dynamic or complex hot water automation (based on Time + upper & lower temperature) - they can use prohibit/boost to trigger an "Eco" hot water run at a given time - for example, some days it might be cheapest at 01:30 to boost HW, but others 03:30

merlin051, 2024-12-03:
I'm coming at it with Octopus Go(00:30-05:30 8.5p/kWh) paired with 30kWh of batteries - slightly different approach in the winter and no viable solar to depend on.

In the summer it's cheaper to charge the tank during the day (usually warmer) and excess solar will be available driven by an automation monitoring the battery %% and the remaining daylight, free beats cheap.

In the winter I get the tank charged and then put as much heat as possible into Zone1 (ground floor stone UFH) so the house fabric soaks it up whilst it's cheap. I've been playing with alternating WC and fixed flow rate with some decent success.

F1p, 2024-12-19:
Original problem should be fixed in v5.3.2 where the mode is kept the same as at the start of the run

closed 2024-12-19 by F1p