F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #9: Offset for Electricity Consumption

merlin051, 2024-12-03:
I think it's well known that the mitsubishi tracked figures are not accurate for power usage.

I've got a CT clamp measuring the HeatPump Consumption and it roughly tracks the same profile, but is way lower on the mitsubishi side when compared to actuals.

Could we have an offset value that can be user defined to make the numbers reflect more accurately?

Here is an image of the Ecodan numbers against a CT clamp which matches what Octopus think I use.
[image]

F1p, 2024-12-03:
What size of outdoor unit do you have?
What is the comparison to Mitsi onboard measurement sensor: "Ecodan ASHP Heat Pump Input Power"?

merlin051, 2024-12-03:
It's an 11.2kw unit.

It looks like the Ecodan values are using floor and ceiling as their are always round so adding an offset here would be difficult I suspect. As it's always going to be +/- .500w

New comparison below 👇 
[image]

F1p, 2024-12-03:
The computed was based orignally on a 8.5kW unit, so under estimate would be expected for a 11.2kW and over estimate for a 6kW units and not accounting for different types of refrigerants.

Of course external CT power monitoring is a much better way to do this, for those who don't have monitoring, this provides some level of estimation
I will look to refine the formula 👍

seumasmorrison, 2024-12-03:
I've noticed that during freezing spells over a 24 period, the CT clamp monitoring almost matches the Ecodan 'yesterday' consumption figure (R290, 6kW).

Not sure if this is just due to the energy used for defrost not being accounted for in the Ecodan estimate.

F1p, 2025-01-01:
Proposal would be either to estimate maximum output power seen, to determine unit size

F1p, 2025-01-06:
@merlin051 
@seumasmorrison 

Please see [Gen2 file for v6.1.1](https://github.com/F1p/Mitsubishi-CN105-Protocol-Decode/blob/master/build/esp32.esp32.m5stack_atoms3/v6.1.1/ECODAN_Bridge_Gen2_v6.1.1.bin) for testing, unit size & glycol strength have been added as controls which impact both the input and output power

seumasmorrison, 2025-01-06:
It correctly defaulted to 8.5 kW (correct for my mother's system) and 30% Glycol, which I guess maybe in the right ball park.

F1p, 2025-01-06:
> It correctly defaulted to 8.5 kW (correct for my mother's system) and 30% Glycol, which I guess maybe in the right ball park.

That's the default starting point, what the calculation was previously based on

merlin051, 2025-01-08:
Will test this at the weekend

merlin051, 2025-01-12:
Ok - updated the firmware and set my punp size to 11.2 and a 20% glycol mix (I'm not 100% sure of my mixture so it could be off) 

On the new charts you can see when the firmware update happened as the energy use steps up indicated by the red line.

It's still slightly off against my CT clamp but much better than it was.
[image]
[image]

F1p, 2025-01-12:
So computed is maybe 200-250W under the CT clamp reading? 
3% @6kW

seumasmorrison, 2025-01-12:
[image]

Time series of Shelly vs Computed input power on 6kW R290, maybe slight under prediction, more noticeable on defrost cycles.

merlin051, 2025-01-12:
[image]

In this screen shot the CT clamp is the top graph and then the computed value is the second.

I patched the new version in at roughly 7:40AM

Before patch power peaks:

CT=> 5709W   Computed=> 3610W Variation=> 2099W
CT=> 5494W   Computed=> 3520W Variation=> 1974W
CT=> 5660W   Computed=> 3590W Variation=> 2070W
CT=> 6565W   Computed=> 4060W Variation=> 2505W

##### Patch Applied ####

After patch power peaks:

CT=> 6614W  Computed=> 6160W Variation=> 454W
CT=> 6270W  Computed=> 5810W Variation=> 460W
CT=> 6384W  Computed=> 5950W Variation=> 434W
CT=> 6303W  Computed=> 5640W Variation=> 663W

It's reading at a much smaller variance in comparison to my CT meter (which tracks pretty closely with the octopus mini when comparing loads)

jeanmoulart, 2025-01-14:
> Proposal would be either to estimate maximum output power seen, to determine unit size or take user input

Hey @F1p, thanks for adding this feature, I had noticed the estimated consumption was significantly higher than the actual one. My outdoor unit has an output of 4kW. Could you add this variant to the drop-down list?

F1p, 2025-01-14:
> Hey @F1p, thanks for adding this feature, I had noticed the estimated consumption was significantly higher than the actual one. My outdoor unit has an output of 4kW. Could you add this variant to [...]

Yes, could you show a comparison of data points to see the difference please?

jeanmoulart, 2025-01-16:
Unfortunately I don't have a specific measurement for the heat pump, only net consumption or net production to the network (from the meter) and current solar production, but I could tell the difference was huge and incoherent before adding the new feature and setting it to 5kW, 0% glycol.

Also, the 4kW output is only a label as the HP can go beyond 4kW when weather conditions are favorable (this is also mentioned in the ecodan documentation), could `sensor.ecodan_ashp_heat_pump_output_power` be used here?

[image]

At last, I have a split unit and refrigerant R32 circulates between the outdoor and indoor unit, not water. I don't know how you do the math here to get an estimate so I can't tell if this could have an impact.

jeanmoulart, 2025-01-20:
I could eventually get some data points, but only for total consumption by comparing : 
   - 1. `Ecodan ASHP Total Consumed Energy Yesterday`
   - 2. An `integration` of `Ecodan ASHP Computed Input Power`

Day 1: 
  1.
  2.
Day 2:
  1.
  2.
Day 3:
1.

F1p, 2025-01-20:
@jeanmoulart - you are running Gen1 device?

You can try pre-release version v6.1.2: https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/tree/master/build/esp8266.esp8266.generic/v6.1.2

jeanmoulart, 2025-01-20:
@F1p I'm running the ethernet device. I have just updated to the latest version. I'll let you know if setting the outdoor unit size to 4kW improves the estimation.

F1p, 2025-01-20:
> [@F1p](https://github.com/F1p) I'm running the ethernet device. I have just updated to the latest version. I'll let you know if setting the outdoor unit size to 4kW improves the estimation.

Great - thank you

F1p, 2025-02-13:
For now, i think we are ok to close following 6.1.2 release?
Other unit sizes will need to be submitted in the future.

merlin051, 2025-02-13:
Using the captured data on an 11.2kw it still looks to be average of 502w off.

I'm not sure if this is something we can tweak further?

I've experimented with all glycol levels and it doesn't appear to have a noticeable impact on the consumption reported.

F1p, 2025-02-14:
> Using the captured data on an 11.2kw it still looks to be average of 502w off.
> 
> I'm not sure if this is something we can tweak further?
> 
> I've experimented with all glycol levels and it doesn't appear to have a noticeable impact on the consumption reported.

502W off in which direction and constantly or in certain situations?

jeanmoulart, 2025-02-14:
Regarding my 4kW installation, I have installed a separate meter to get better power consumption data. It turns out that the estimation provided by `Ecodan ASHP Computed Input Power` with 5kW unit and 0% glycol as parameters is pretty close to what's measured by the meter : 

(sorry I could not display both in the same chart)
[image]

F1p, 2025-02-19:
> Regarding my 4kW installation, I have installed a separate meter to get better power consumption data. It turns out that the estimation provided by `Ecodan ASHP Computed Input Power` with 5kW unit [...]

Maybe 4kW and 5kW unit share the same compressor size, so get similar values