F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #12: Zone 1 heating

Callumhome, 2024-12-22:
Have discovered a issue with latests firmware when running the zone one heating which triggers by automation it will change on thermostat to heating but remains greyed out so doesn't heat look into it come to heat then back to off

F1p, 2024-12-22:
This is latest v5.3.2?

Do you have any other automations triggering at the same time to the heat pump?

Callumhome, 2024-12-22:
No other automations with this i only have heat or or heat off automation and this is latest version had to downgrade because it should kicked the heating at 7 am it went on but it will fully grayed out

F1p, 2024-12-22:
> No other automations with this i only have heat or or heat off automation and this is latest version had to downgrade because it should kicked the heating at 7 am it went on but it will fully grayed out

Using server control mode and Prohibit?
If it was greyed out had it exited server control?

Callumhome, 2024-12-22:
Yeah I use server controll and then unprohibit heating when required

F1p, 2024-12-27:
Fixed in v6.0.0 with command queue