F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #13: invalid action_template

musapinar, 2024-12-31:
Hello,

Can we fix this please ?

```
Error 'invalid template (TemplateSyntaxError: Encountered unknown tag 'else'.) for dictionary value @ data['action_template']' when processing MQTT discovery message topic: 'homeassistant/climate/fc525edab734du/config', message: '{'device': {'identifiers': 'EcodanBridge-fc525edab734'}, 'unique_id': 'ashp_Zone2_flow_climate_fc525edab734', 'object_id': 'zone2_flow_climate', 'current_temperature_topic': 'Ecodan/ASHP/Status/System', 'current_temperature_template': '{{ value_json.HeaterFlow|round(1) }}', 'temperature_command_topic': 'Ecodan/ASHP/Command/Zone2/FlowSetpoint', 'temperature_state_topic': 'Ecodan/ASHP/Status/Zone2', 'temperature_state_template': '{{ value_json.FSP }}', 'temperature_unit': 'C', 'temp_step': 0.5, 'precision': 0.5, 'initial': 30, 'action_topic': 'Ecodan/ASHP/Status/System', 'action_template': "{%set mode=value_json.SystemOperationMode|lower%}{%set is_working=states('sensor.ecodan_ashp_zone_2_working')|bool%}{%set is_prhbt=states('sensor.ecodan_ashp_zone_2_heating_prohibit')==1 or states('sensor.ecodan_ashp_zone_2_cooling_prohibit')==1%}{%if mode=='off'%}{%if not is_prhbt and not is_working%}idle{%else%}off{%endif%}{%elif is_working%}{{mode}}{%elif not is_prhbt%}idle{%else%}off{%endif%}{%else%}off{%endif%}", 'mode_state_topic': 'Ecodan/ASHP/Status/System', 'mode_state_template': "{{'heat' if (value_json.OpMode=='Heat' and state_attr('climate.zone2_climate','current_temperature')!=0 and states('sensor.ecodan_ashp_zone_2_heating_prohibit')=='0') else 'cool' if (value_json.OpMode=='Cool' and state_attr('climate.zone2_climate','current_temperature')!=0 and states('sensor.ecodan_ashp_zone_2_cooling_prohibit')=='0') else 'off'}}", 'mode_command_template': "{%set values = {'heat':'Heating Temperature','cool':'Cooling Temperature'}%}{{values[value] if value in values.keys()}}", 'mode_command_topic': 'Ecodan/ASHP/Command/System/HeatingMode', 'availability': {'topic': 'Ecodan/ASHP/Status/Zone2', 'value_template': "{{'online' if value_json.HeatingControlMode in ['Flow','Cooling Flow'] else 'offline'}}"}, 'modes': ['heat', 'cool', 'off'], 'min_temp': 10, 'max_temp': 60, 'name': 'Zone 2 Flow Thermostat'}'
```

[image]

There seems to be an error with the nested ifs in the action_template ?

This is the current state :
```
{%set mode=value_json.SystemOperationMode|lower%}
{%set is_working=states('sensor.ecodan_ashp_zone_2_working')|bool%}
{%set is_prhbt=states('sensor.ecodan_ashp_zone_2_heating_prohibit')==1 or states('sensor.ecodan_ashp_zone_2_cooling_prohibit')==1%}
{%if mode=='off'%}
  {%if not is_prhbt and not is_working%}
    idle
  {%else%}
    off
  {%endif%}
{%elif is_working%}
  {{mode}}
{%elif not is_prhbt%}
  idle
{%else%}
  off
{%endif%}
{%else%}
  off
{%endif%}
```
which seems wrong to me.

This is what it would expect it to look like : 
```
{% set mode = value_json.SystemOperationMode | lower %}
{% set is_working = states('sensor.ecodan_ashp_zone_2_working') | bool %}
{% set is_prhbt = (states('sensor.ecodan_ashp_zone_2_heating_prohibit') == '1' or states('sensor.ecodan_ashp_zone_2_cooling_prohibit') == '1') %}
{% if mode == 'off' %}
    {% if not is_prhbt and not is_working %}
        idle
    {% else %}
        off
    {% endif %}
{% elif is_working %}
    {{ mode }}
{% elif not is_prhbt %}
    idle
{% else %}
    off
{% endif %}
```

inline : 
```
"action_template": "{% set mode = value_json.SystemOperationMode | lower %}{% set is_working = states('sensor.ecodan_ashp_zone_2_working') | bool %}{% set is_prhbt = (states('sensor.ecodan_ashp_zone_2_heating_prohibit') == '1' or states('sensor.ecodan_ashp_zone_2_cooling_prohibit') == '1') %}{% if mode == 'off' %}{% if not is_prhbt and not is_working %}idle{% else %}off{% endif %}{% elif is_working %}{{ mode }}{% elif not is_prhbt %}idle{% else %}off{% endif %}",
```

Thank you

F1p, 2024-12-31:
Looks ok in current commit - maybe needs recompiled?

https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/blob/d8b374bbe07eaba1f92b23b7d7d5d65de47255b3/MQTTDiscovery.h#L587

`{%set mode=value_json.SystemOperationMode|lower%}
{%set h_prhbt=not(states('sensor.ecodan_ashp_zone_2_heating_prohibit')|bool)%}
{%set c_prhbt=not(states('sensor.ecodan_ashp_zone_2_cooling_prohibit')|bool)%}
{%set is_wrk=not(states('sensor.ecodan_ashp_zone_2_working')|bool)%
{%if mode=='off' and (h_prhbt or c_prhbt) and (zone_temp==0 or is_wrk)%}idle
{%elif is_wrk and (h_prhbt or c_prhbt)%}idle
{%else%}
{{mode}}
{%endif%}`

musapinar, 2025-01-01:
Does the OTA update via the web UI work (http://192.168.x.x/update) ?
In case it does, which file do I have to upload ? I guess the one ending in ".merged.bin" ?

F1p, 2025-01-01:
> Does the OTA update via the web UI work (http://192.168.x.x/update) ? In case it does, which file do I have to upload ? I guess the one ending in ".merged.bin" ? [...]

Yes the OTA update is how to do it. 
Use the .bin file for your hardware type.

Latest releases here: https://github.com/F1p/Mitsubishi-CN105-Protocol-Decode/releases/tag/v6.0.0

You are selecting 6.1.0 which is not a formal release yet and may have other bugs

musapinar, 2025-01-01:
I had version 5.3.2.
6.0.0 now. Update ran successfully.
Thank you.

closed 2025-01-01 by musapinar