F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #16: Warnings in the logs

jeanmoulart, 2025-01-30:
Hello @F1p,

I realized I had some warnings in my log and decided to do some cleaning and remove the old entities coming from before the renaming to `ecodan_ashp_...`. I removed them in `config/.storage/core.entity_registry' and restarted HA. Most of them disapparead but still 5 of them came back and I still have this warning.
`
`2025-01-30 13:33:11.648 WARNING (MainThread) [homeassistant.components.mqtt.entity] TypeError: '>' not supported between instances of 'NoneType' and 'int' rendering template for entity 'climate.zone1_flow_climate', template: '{%set mode=value_json.SystemOperationMode|lower%}{%set zone_temp=state_attr('climate.zone2_climate','current_temperature')%}{%set h_prhbt=not(states('sensor.ecodan_ashp_zone_1_heating_prohibit')|bool)%}{%set c_prhbt=not(states('sensor.ecodan_ashp_zone_1_cooling_prohibit')|bool)%}{%set is_wrk=not(states('sensor.ecodan_ashp_zone_1_working')|bool)%}{%if mode=='off' and (h_prhbt or c_prhbt) and (zone_temp==0 or is_wrk)%}idle{%elif zone_temp>0 and is_wrk and (h_prhbt or c_prhbt)%}idle{%else%}{{mode}}{%endif%}' and payload: {"HeaterFlow":49,"HeaterReturn":47,"FlowReturnDeltaT":2,"OutsideTemp":7,"Defrost":"Off","InputPower":1,"HeaterPower":2,"EstInputPower":1.13,"EstHeatOutputPower":2.79,"EstCoolOutputPower":0,"Instant_CoP":2.47,"Compressor":48,"SystemPower":"On","SystemOperationMode":"Hot Water","HolidayMode":0,"FlowRate":20,"RunHours":1660,"HB_ID":35}
2025-01-30 13:33:11.648 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'HeaterSetpoint' when rendering '{{ value_json.HeaterSetpoint }}'
2025-01-30 13:33:11.649 WARNING (MainThread) [homeassistant.components.mqtt.climate] Invalid ['cooling', 'defrosting', 'drying', 'fan', 'heating', 'idle', 'off', 'preheating'] action: hot water
2025-01-30 13:33:11.649 WARNING (MainThread) [homeassistant.components.mqtt.climate] Invalid ['cooling', 'defrosting', 'drying', 'fan', 'heating', 'idle', 'off', 'preheating'] action: hot water`

Any idea how I could get rid of the warnings?

Thanks!

F1p, 2025-01-30:
Are you operating with a single zone?

Errors can occur when zone2 entites are disabled, there could be a to-do/improvement item to use dip-switches to determine which entities are approrirate for the user

jeanmoulart, 2025-01-30:
Yes, I'm only operating with a single zone.

F1p, 2025-01-30:
> Yes, I'm only operating with a single zone.

Have you disabled Zone2 Temperature Climate & Zone2 Flow Climates?

jeanmoulart, 2025-01-30:
Looks like I haven't

[image]

F1p, 2025-01-30:
Looks like your Zone2 Thermostat is called: "climate.zone2_climate_2" rather than "climate.zone2_climate" which it is expecting
Do you have another climate.zone2_climate?

jeanmoulart, 2025-01-30:
I removed two entities : `climate.zone1_climate_1` and `climate.zone2_climate_2`

jeanmoulart, 2025-01-30:
I removed two entities : `climate.zone1_climate_1` and `climate.zone2_climate_2`

No more warnings, thanks.

closed 2025-01-30 by jeanmoulart