F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #18: Compilation error

PeterPrelo, 2025-03-22:
C:\Users\HP\Documents\Arduino\Projekti\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\ECODAN_Bridge\ECODAN_Bridge.ino:254:53: error: 'FTCCable_RxPin' was not declared in this scope
  254 |   HEATPUMP_STREAM.begin(SERIAL_BAUD, SERIAL_CONFIG, FTCCable_RxPin, FTCCable_TxPin);  // Rx, Tx
      |                                                     ^~~~~~~~~~~~~~
C:\Users\HP\Documents\Arduino\Projekti\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\ECODAN_Bridge\ECODAN_Bridge.ino:254:69: error: 'FTCCable_TxPin' was not declared in this scope
  254 |   HEATPUMP_STREAM.begin(SERIAL_BAUD, SERIAL_CONFIG, FTCCable_RxPin, FTCCable_TxPin);  // Rx, Tx
      |                                                                     ^~~~~~~~~~~~~~
C:\Users\HP\Documents\Arduino\Projekti\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\ECODAN_Bridge\ECODAN_Bridge.ino:256:48: error: 'MEL_RxPin' was not declared in this scope
  256 |   MEL_STREAM.begin(SERIAL_BAUD, SERIAL_CONFIG, MEL_RxPin, MEL_TxPin);  // Rx, Tx
      |                                                ^~~~~~~~~
C:\Users\HP\Documents\Arduino\Projekti\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\ECODAN_Bridge\ECODAN_Bridge.ino:256:59: error: 'MEL_TxPin' was not declared in this scope
  256 |   MEL_STREAM.begin(SERIAL_BAUD, SERIAL_CONFIG, MEL_RxPin, MEL_TxPin);  // Rx, Tx
      |                                                           ^~~~~~~~~
C:\Users\HP\Documents\Arduino\Projekti\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\ECODAN_Bridge\ECODAN_Bridge.ino:265:11: error: 'Reset_Button' was not declared in this scope
  265 |   pinMode(Reset_Button, INPUT);  // Pushbutton on other modules
      |           ^~~~~~~~~~~~
C:\Users\HP\Documents\Arduino\Projekti\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\ECODAN_Bridge\ECODAN_Bridge.ino: In function 'void loop()':
C:\Users\HP\Documents\Arduino\Projekti\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT-master\ECODAN_Bridge\ECODAN_Bridge.ino:457:19: error: 'Reset_Button' was not declared in this scope
  457 |   if (digitalRead(Reset_Button) == LOW) {                                                                                                                                                                    // Inverted (Button Pushed is LOW)
      |                   ^~~~~~~~~~~~

exit status 1

Compilation error: 'FTCCable_RxPin' was not declared in this scope

F1p, 2025-03-22:
What board and what Arduino core are you compiling for?

PeterPrelo, 2025-03-22:
I am building with the ESP32-S3-Zero board and using Arduino Version: 2.3.4

F1p, 2025-03-22:
This board version:
[image]

PeterPrelo, 2025-03-22:
[image]
Board version

F1p, 2025-03-22:
Try 3.1.1?

F1p, 2025-03-22:
https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/blob/8f2bd0fd1be8388ea892463acadb768359139faa/ECODAN_Bridge.ino#L76

Change this line to ESP32, because pins and features for different hardware types

PeterPrelo, 2025-03-22:
I had to replace ARDUINO_M5STACK_ATOMS3 with ESP32 everywhere

F1p, 2025-03-22:
> I had to replace ARDUINO_M5STACK_ATOMS3 with ESP32 everywhere because the ESP32-S3 Zero has a built-in RGB LED

Okay, hope you enjoy!

PeterPrelo, 2025-03-22:
I am very grateful for the effort you put into developing this program.
Thank you again.

closed 2025-03-22 by F1p