F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #19: Using ASHP input power in the energy dashboard

mr-miles, 2025-04-02:
Hi

I would like to add the heatpump as a specific device in the homeassistant energy dashboard for a rough estimate of energy usage.

I've added the "integral" integration and pointed at the "computed total input power" sensor, and it created a sensor that is generating some figures.

However, the energy dashboard configuration page complains that this sensor doesn't have "device class: energy" associated, but it should be added automatically.  If I add "device class: power" to the "computed <x> input power" sensors that are auto-setup, the message goes away.

Is it possible to add that device class power attribute so the "computed <x> input power" sensors get it when they are discovered?

Thanks for your help!

Miles

F1p, 2025-04-02:
Energy sensors need to be in the format: Select the energy sensor that measures the device's energy usage in either of cal, Gcal, GJ, GWh, J, kcal, kJ, kWh, Mcal, MJ, MWh, mWh, TWh, Wh.
Therefore i'm hestitant if it would be suitable to put  "device_class": "power" against Computed Input & Outputs

In the [documentation](https://www.home-assistant.io/integrations/integration/#energy) it will point you to using the Integral Helper as suggested.

Although i have the same warnings about device class, the individual devices energy section still populates:
[image]

[image]

mr-miles, 2025-04-02:
Sorry, I just meant to put it against the sensors that measure in kW (power sensors) rather than the kWh ones (which are energy sensors and come up no problem in the energy dashboard).

The integral helper will put "device_class: energy" on the sensor it creates (which makes the warning go away), IF the sensor it is based on has "device_class: power" as an attribute - https://github.com/home-assistant/core/blob/a7be9e664356e6ee496c3cb6169dcf39f0964dec/homeassistant/components/integration/sensor.py#L85 - and then there is no warning at all.

I saw some sensors already declare a device_class: energy in the discovery doc and so was thinking this would be a handy addition (at that point in the day I was confused as the dashboard value was stuck on zero - but this turned out to be due to the total usage chart only updating hourly.)

F1p, 2025-04-04:
Added as requested in Commit [a8e9f52](https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/commit/a8e9f52913de61d1008c6771122425c12cbfdbff)

You can test pre-release by uploading from the [/build](https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/tree/master/build) folder: [ECODAN_Bridge_Gen2_v6.2.3.bin](https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/blob/master/build/esp32.esp32.m5stack_atoms3/v6.2.3/ECODAN_Bridge_Gen2_v6.2.3.bin)

Note, this compile also bumps ESP to [Arduino Core 3.2.0](https://github.com/espressif/arduino-esp32/releases/tag/3.2.0) which needed a little modification for DeviceID, so full release will be a little delayed to allow for testing. Any new issues related to this please assist with a new issue

Thanks very much!
Phil

mr-miles, 2025-04-04:
Thanks, I am running this now.

closed 2025-06-20 by F1p