F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #21: Can't set HotWater SetPoint

tituskurth, 2025-05-02:
hi there.

I stumbled over a Problem: 
I can not set the HotWater SetPoint. I send it, i see the set command in the mqtt, also in inside the status i see it, but it get's reset after a about 30-40 seconds to the old value. The same happens on System Power and also SvrControlMode.

FlowSetpoint of Zone1 works without any Problems, also all most readings works without Problem.

The debug output from setting HotWater SetPoint to 45:

`Received MQTT Message on topic Ecodan/ASHP/Command/HotWater/Setpoint with Payload 45
MQTT Set HW Setpoint
Transferred msg to position: 1
Stopping Heat Pump Read Operation to FTC
Writing msg at position: 1, attempt: 1
[Bridge > FTC] fc, 41, 02, 7a, 10, 32, 20, 00, 00, 00, 00, 00, 00, 11, 94, 00, 00, 00, 00, 00, 00, 3c, 
[FTC > Bridge] fc, 61, 02, 7a, 10, __, __, __, __, __, __, __, __, __, __, __, __, __, __, __, __, 13, 
Write OK!
MQTT Published!
`

The WiFiStatus output:

`{
  "SSID": "FRITZ!Box",
  "RSSI": -54,
  "IP": "192.168.1.111",
  "Firmware": "6.2.3-h1",
  "CPUTemp": 77.22,
  "CPULoopTime": 1961,
  "FTCLoopTime": 19856,
  "FTCReplyTime": 126,
  "FTCVersion": "FTC4",
  "FTCSoftwareVersion": "12.01",
  "FTCTime": "2025-05-02T18:41:40Z",
  "UnitSize": "8.5",
  "Glycol": "30%",
  "HB_ID": 56
}`

Is there any known limitations for this FTC Version?
Is there a way to set the Hotwater Temp?
Main reason is to bring this to heat in combination with a Photovoltaic system.

Thanks for your help.

tituskurth, 2025-05-02:
Thanks for the Quick response.
I didn't know this docs, but seems helpful.
So i can't set the Tank Temp because it's read only for me.
Server Control Mode is also not possible, it gets also reset, same with any boost commands.
I also can't switch it off with the POWER command.
I tried also Firmware 6.1.2 but nothing changed so far.

What i don't understand: I can change this temp over the FTC4 directly and also with MelCloud, so this should not be read only.

But anyway, this seems not a solution, in no way :(

F1p, 2025-05-02:
I worked with a user before, probably September last year and force DHW, holiday mode and setting zone temperatures and On/Off worked fine.

Server control we saw on main controller it started, but would exit like you say.

Can you observe main controller and perform commands?

tituskurth, 2025-05-02:
yes, i stand in front of the main controller and issue the commands, but it changes nothing, except on changing Flow Setpoint of Zone 1, that worked immediately.

F1p, 2025-05-02:
These other commands work from MELCloud ok?

tituskurth, 2025-05-02:
i used only the Flow Setpoint of Zone 1 and Tank Temp setting, both worked fine, until they kill and block the heatpunk/account as a result over "overusage"

F1p, 2025-05-02:
Did you get a board to allow MELCloudto run as well?
Perhaps we can record & compare the commands sent

tituskurth, 2025-05-02:
no, sorry, no board.
Is it 1:1 wired or are there other components on this board?

F1p, 2025-05-02:
Okay, maybe we can consider it - it works like so:
[image]

tituskurth, 2025-05-02:
i have checked also this one: https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CNRF
`{
  "QTY": 2,
  "Power": "Standby",
  "SystemOperationMode": "Off",
  "Zone1OperationMode": "Heating Auto Adapt",
  "Zone2OperationMode": "Heating Auto Adapt",
  "TimerProhibit": "None",
  "Z1ActiveInput": 0,
  "Z2ActiveInput": 0,
  "Z1Setpoint": 0,
  "Z2Setpoint": 0,
  "ErrorCode": 0,
  "DHWForce": "Off",
  "Holiday": "Off",
  "HolidayHrs": 0,
  "RC1Input": 18,
  "RC2Input": 18,
  "RC3Input": 18,
  "RC4Input": 18,
  "RC5Input": 18,
  "RC6Input": 18,
  "RC7Input": 18,
  "RC8Input": 18,
  "HB_ID": 40
}
`

but this not even give me the heater temp :(

so there seems not really a solution for me

F1p, 2025-05-02:
Send me an email to arrange proxy board to run MELCloud in parallel

closed 2025-05-30 by F1p