F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #22: v6.2.4 binaries corrupted

musapinar, 2025-06-05:
Latest bins do not seem to work. Flashing OTA through the web UI returns errors :

with ECODAN_Bridge_Gen2_v6.2.4-h1.bin
```
Update failed!
Reboot device and try again
OTA Error: Flash Read Failed
```

with ECODAN_Bridge_Gen2_v6.2.4.bin
```
Update failed!
Reboot device and try again
OTA Error: Decryption error
```

this happens flashing via the WLAN (192.168.x.x), I did not try via the ESP's own AP (aka "captive portal").

current fw is 6.1.1 (Build date Jan 9 2025 10:27:41)

musapinar, 2025-06-05:
M5Stack dev kit

F1p, 2025-06-05:
Is it possible to do progressive updates in smaller steps?
6.1.1 to 6.2.0
6.2.0 to 6.2.4-h1?

It may be that the file size is too large in most recent and partition scheme needs resized (clean flash with .merged.bin) as described [here](https://github.com/F1p/Mitsubishi-CN105-Protocol-Decode/blob/master/documentation/Hardware%20and%20Recovery.md)

closed 2025-07-16 by F1p