F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #24: Error in logs when heating water

gracar2002, 2025-07-07:
Hardware version: Ethernet without passthrough
Firmware version: 6.2.4

It appears that when the DHW is running an invalid climate action is being passed through which causes the below error to appear in homeassistant

`Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:692
integration: MQTT (documentation, issues)
First occurred: 05:01:54 (858 occurrences)
Last logged: 05:28:08

Invalid ['cooling', 'defrosting', 'drying', 'fan', 'heating', 'idle', 'off', 'preheating'] action: hot water`

F1p, 2025-07-07:
DHW Climate should display "heating" when operating as per [L679](https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/blob/3cbefea73bd370bd38c8f9bb1f36cd52b79194ae/MQTTDiscovery.h#L679)

Thermostat & Flow Climate entities should show "idle" or "off" as per [L680 to L683](https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/blob/3cbefea73bd370bd38c8f9bb1f36cd52b79194ae/MQTTDiscovery.h#L680)

Manually testing seemed to show a valid response:

[image]

Can you check your retained topic:
homeassistant > climate > <deviceID>cf/cg/dw/du > action_template?

[image]

gracar2002, 2025-07-07:
Upgrade done within a couple of days of 6.2.4 being released and had multiple restarts since then - Bridge is not in server mode if that's any help?

gracar2002, 2025-07-08:
Appears to have resolved by deleting topic (once I remembered you need to send a command to the ecodan to get the full publish)

closed 2025-07-08 by gracar2002

F1p, 2025-07-08:
Great, thanks for the update - I suspect it was a previous config which didn't include hot water and it was still floating around somewhere in home assistant or the MQTT broker