F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #29: Compensation Curve

merlin051, 2025-09-09:
I'm not seeing the documented behaviour, specifically when activating the compensation curve it doesn't seem to set the mode to 'Fixed Flow'

`By activating the onboard compensation curve, if not already in Fixed Flow the software will change your Ecodan to this mode.`

My yaml has been accepted for setting up the compensation curve and I can see it in the status topic when I configure a listener.

Is there something wrong with my activation yaml below?

`sequence:
  - action: mqtt.publish
    data:
      evaluate_payload: false
      qos: 0
      topic: Ecodan/ASHP/Command/System/CompCurve
      payload: |-
        {   
            "zone1": {
                "active": true
            },
            "zone2": {
                "active": true
            }
        }
alias: Enable HeatPump Compensation Curve
description: ""`

F1p, 2025-09-10:
> Worked once, back to doing what it did previously

Can you just confirm the order of operations are as follows:

- Command/System/CompCurve to Disable e.g.: `{"zone1": {
        "active": false
    }
}`
- (Confirm Status shows correctly as disabled)
- Change your Operation Mode Selector to "Heating Compensation"
-  Command/System/CompCurve to Enable e.g.: `{"zone1": {
        "active": true
    }
}`
- (Confirm Status shows correctly as enabled)
- On transition from Active "False" to "True", if your System Operation mode for the Zone was not Fixed Flow it should of changed

F1p, 2025-09-10:
> 1. The `{"zone1": { "active": false } }` was sent, I assumed it would default the heating mode back away from Fixed Flow , this was not observed, so I added a task to my script to do this.

It was decided not to save the mode state on entry or revert back and perform the action automatically - it was only to ensure the user was operating in Fixed Flow for the feature to work

> 2. Looking at the MQTT listener the status is showing as enabled  (active: true)

I tested with the flow above and mine was exiting correctly when "false" was sent for either/both zones. 
I had the same issue entering last night.

> 3. Flipping the heating to 'Heating Temperature' then sending the activation message does nothing, the mode does not flip to 'Heating Flow'

This is expected, because there is no transition from active:false to active:true if it was already true as per point no. 2

> 4. its like any active=false/active=true is being ignored now.

It sounds like this, yes, but i wasn't able to replicate (yet)

> 5. Setting the offsets still works and is reported on the mqtt listener

Yes, this should operate regardless of active/inactive

closed 2025-09-18 by F1p

merlin051, 2025-09-18:
We're you able to replicate this and make an adjustment? (I haven't had a chance to go back to investigating yet)

F1p, 2025-09-18:
Yes, and another report merged in
Would be good to hear when you have a chance

F1p, 2025-09-18:
> We're you able to replicate this and make an adjustment? (I haven't had a chance to go back to investigating yet)

And of course, thank you very much for supporting testing & reporting this back to help improve the feature

F1p, 2025-09-22:
Re-opened

You have a typo in your disable JSON:

`{
"zone1": {
"active": false
},
"zone2: {
"active": false
}
}`

You are missing the close quote on Zone2, it should be:
`{
"zone1": {
"active": false
},
"zone2": {
"active": false
}
}`

F1p, 2025-09-22:
> activation was detected regardless of existing state

Change to Fixed Flow is on transition only, so currently it is operating as designed