F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #31: Computed Input Power Logic

nildan, 2025-09-17:
Following up on #23, I can confirm that everything has been working fine for a while now. Thanks for a great product :)

I still notice a detail concerning DHW though, please see the history graphs from HA below. Right at the beginning of a DHW cycle, the input power is attributed to Cooling, since DeltaT is negative. My heat pump is never in Cooling mode, but maybe this logic is intented?

Then, Computed DHW Input Power (blue line in the first graph) is zero briefly at two moments (when DeltaT is zero). However, Total Input Power (yellow line) is still >1 kW during these moments. This power is not attributed to Cooling or Heating either.

This is only for 1-2 minutes, so it won´t affect Energy figures much, but maybe points to something in the logic.

[image]

F1p, 2025-09-17:
Are you running bridge firmware version 6.3.x?

nildan, 2025-09-17:
Yes, 6.3.3.

F1p, 2025-09-17:
> Yes, 6.3.3.

Proposed fix in 6.3.5:
[Gen2 update file](https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/blob/master/build/esp32.esp32.m5stack_atoms3/v6.3.5/ECODAN_Bridge_Gen2_v6.3.5.bin)

nildan, 2025-09-17:
> > Yes, 6.3.3.
> 
> Proposed fix in 6.3.5: [Gen2 update file](https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/blob/master/build/esp32.esp32.m5stack_atoms3/v6.3.5/ECODAN_Bridge_Gen2_v6.3.5.bin)

Great, updated now and will report back after next DHW cycle.

F1p, 2025-09-17:
> > > Yes, 6.3.3.
> > 
> > 
> > Proposed fix in 6.3.5: [Gen2 update file](https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/blob/master/build/esp32.esp32.m5stack_atoms3/v6.3.5/ECODAN_Bridge_Gen2_v6.3.5.bin)
> 
> Great, updated now and will report back after next DHW cycle.

Please do - thanks!

nildan, 2025-09-17:
It no longer attributes power to Cooling - great!

DHW Input Power still goes to zero. It seems that Input Power can be positive even though DeltaT and Output Power is zero. When looking at the code I am guessing that the case when OutputPower = 0 is not covered by the logic, only <0 and >0, so Input Power during those moments is not attributed to any mode.

It happens for 20 seconds and corresponds to 0.007 kWh though, so not a big deal :)

[image]

nildan, 2025-09-18:
> Perhaps the pump hadn't started yet - what was the value of Compressor Frequency in this graph?

Compressor Frequency looks like this:

[image]

F1p, 2025-09-18:
> Ah, yes. What I meant was these approx 4+16 seconds, where total Input Power (blue line) continues but DHW Input Power (yellow line) drops to zero.
> 
> [image]
> 
> This happens when DeltaT = 0 and thus OutputPower = 0.

Ah sorry! My misunderstanding
Yes - [L1070](https://github.com/F1p/Home-Assistant-Mitsubishi-Ecodan-CN105-to-MQTT/blob/853d50d24bc58252f7aac33942a7b5a5ffebf247/ECODAN_Bridge.ino#L1070) should be less than or equal to 0

Will fix this

nildan, 2025-09-19:
It's working perfectly now, thanks for fixing it so quickly!

[image]