F1p/Home-Assistant-Mitsubishi-CN105-to-MQTT #34: Weather Comp - Temp Offset

merlin051, 2025-09-22:
I've got a script task to set the temp_offset to -2 (for overnight) called from an automation.

However I can't seem to set it back to zero, I can set it to other values without issue, it's like the zero is being ignored.

My idea is that I'll have an automation turning the offset on and off in line with overnight periods as a set back which the standard controller doesn't do in WC mode.

I've tried:

1 : works
0 : doesn't work
-0 : doesn't work 

`{   
    "zone1": {
        "manual_offset": 0,
        "wind_offset": 0,
        "temp_offset": 0
  }
}`

F1p, 2025-09-22:
Same issue as before, it's part of the check to see fields of the JSON payload were sent and "0" is thinking its empty...
Fix incoming

merlin051, 2025-09-22:
Confirmed working in 6.3.5-h2